snoplus/orca #10: 'HV panic down' and 'All triggers off' buttons: global, big and obvious

fbdescamps, 2016-01-14:
It should be very easy to find the HV panic-down and the all-triggers-off buttons. 
We discussed two solutions:
 * Have a separate GUI that always stays on top with the two big buttons in obvious colors. This GUI can then be placed in one corner. 
 * Have these buttons inside a footer in every GUI element.
The first option is probably least confusing.

tlatorre-uchicago, 2016-05-03:
Mostly solved by Javi's new SNO+ GUI. Should we close? @marzece found a way to make sure that this window always stays on top, so we could wait for that. Opinions?

fbdescamps, 2016-05-03:
You mean to leave the SNO+ GUI always on top? That could be a bit cumbersome (depending on screen resolution :P). I still think it would be a good thing to have a dedicated smaller window for this. Other opinions?

marzece, 2016-05-03:
Yeah the floating on top thing is pretty easy. I was considering either doing it to the whole SNOP GUI and having an itunes style shrink window button. Or have a button that pops up a window that floats on top and has the "danger zone" buttons. The first has the advantage of not making duplicate buttons but I'm less sure how to implement it.

fbdescamps, 2016-08-15:
After discussion in the DAQ workshop on August 15th, I think it was decided that we would go with a dedicated  GUI that always stays on top with the two big buttons in obvious colors. This GUI can then be placed in one corner and persist over different desktops.  Volunteers?

fbdescamps, 2016-08-29:
Noel is standing next to me and confirms that this would be nice to have.

fbdescamps, 2017-01-12:
I am going to close this issue. If operator feedback indicates that we need an 'always-on-top' GUI element for this, we can re-open.

closed 2017-01-12 by fbdescamps