snoplus/orca #17: Need to be able to query the trigger status for all channels

marzece, 2016-01-18:


tlatorre-uchicago, 2016-05-04:
Partially mitigated by the detector state database. Should we keep this open?

fbdescamps, 2016-08-17:
It depends if ORCA needs to know about this? I agree that this is now added to the detectorDB. @marzece @tlatorre-uchicago : I feel this issue can be closed if you agree.

marzece, 2016-08-17:
I think it'd still be nice to have if for no other reason than it would be nice to have while debugging. If this doesn't exist than the only way to learn about the state would be to start a run and then go look at the monitoring web page, or to send the XL3 server a save command and manually look through the DB, both are a pretty big pain.

Can/Does the XL3 server server provide a function to send back the state of the N100/N20 triggers? If so we could add to orca a "get triggers" button that will update the gui to reflect what the XL3 server tells it.

tlatorre-uchicago, 2016-08-22:
The XL3 server currently doesn't provide a function to return the trigger state.

marzece, 2016-08-22:
But it could without much issue, right? It must know the trigger state well enough to stuff it into the DB. Is there  any reason we couldn't add something to read that info for non-DB purposes as well.

tlatorre-uchicago, 2016-08-22:
Yeah, although the return format will need to be thought out. The redis protocol supports lists and dictionaries, but I don't think we implement either of those in the redis object in ORCA.

marzece, 2016-08-23:
Probably it'd be best to just dodge that issue and just return a given fecs n100/n20 mask using a 32-bit integer. Unless you're just itching to add list support of course, in which case go for it.

fbdescamps, 2016-10-20:
@tlatorre-uchicago @marzece Is there an update on this issue? How do we resolve it?

tlatorre-uchicago, 2016-10-20:
I think it should be closed. It's not just the triggers, but all the hardware settings that ORCA doesn't know about. The current solution is to either:

1. Start a new run and check the monitoring page
2. Do a cold start/stop which will automatically make the GUI state = hardware state

The only alternative is to have ORCA know the full detector state, which is not in the current plans. We have discussed having a global "SYNC GUI/HW" state button. That should not be too incredibly difficult once we have the postgres integration ironed out.

If it turns out that knowing the triggers is especially important to know while debugging and starting a new run is proving to be an annoyance, we could do as Eric suggested and special case the triggers by adding a command to the XL3 server and a "SYNC triggers with HW" button in ORCA, but it feels like a band aid.

marzece, 2016-10-21:
This is still something I'd like to see, but I'm fine with this issue being closed since the detector state DB makes finding out the triggers doable, if not entirely painless.
I'll just keep it as a personal issue.

tlatorre-uchicago, 2016-11-30:
You will be able to do this with the following query when the next DAQ release comes out:

# SELECT * FROM current_detector_state WHERE crate = 19 AND slot = 10;

In addition, Phil is working on implementing the SYNC GUI/HW button.

closed 2016-11-30 by tlatorre-uchicago