snoplus/orca #28: delete incomplete GUI elements

tlatorre-uchicago, 2016-02-01:
There are a lot of GUI elements that look like they were started by someone, but never finished. For example:

- detector tab in SNO+ window
- details tab in SNO+ window
- HW Map tab in SNO+ window
- database buttons in MTC GUI
- trigger zero finder in MTC GUI

tlatorre-uchicago, 2016-04-13:
Details and HW Map tab have been deleted in #55.

fbdescamps, 2016-12-11:
We should also delete, if we can:
 * data explorer
 * header explorer
 * any of the data-handling stuff

Specifically, we should try to get rid of the following warning:
``` Warning: Run started with empty readout list```

tlatorre-uchicago, 2016-12-14:
I don't think it's possible to get rid of that warning, because the Run Control has to be hooked up to the readout list and that is empty for us since ORCA isn't handling the data.

tlatorre-uchicago, 2016-12-14:
detector tab in SNO+ window was deleted in #177

marzece, 2016-12-14:
We should delete the most amount of stuff possible out of the Experiment though. I don't think we can get rid of the read out list and definitely not the run control but certainly the data explorer and monitor and stuff can go. And all those scripts that are floating around should be copy and pasted to some files and saved somewhere, then deleted I think. I'd be happy to do it on my next shift. If there's no objections

tlatorre-uchicago, 2016-12-17:
I deleted all of the unnecessary objects from the Experiment window.

fbdescamps, 2016-12-18:
Awesome, thanks! Shall we leave this issue or retire it?

tlatorre-uchicago, 2016-12-18:
There are still a few things left:

1. The database buttons and trigger finder panel in the MTC GUI (I think @marzece is working on this)
2. The PMT online/offline switches
3. The fourth PMT state box
4. The database tab on the FEC GUI (currently doesn't save anywhere)

I think that's it.

fbdescamps, 2017-01-05:
Any volunteers to work on these last remaining items? 🐺 🏒 👍

marzece, 2017-01-09:
Do we want to delete all or some of the MTC info from the "Standard Ops" tab. i.e the GTID the 10Mhz Count the GTID rate. I know a lot of it in theory could be grabbed from the MTC server. But I think a lot of that info is pretty useless or doesn't belong in orca. 

The only ones I can imagine might be worth keepings is the rate and the GTID. But I'd still rather just delete them.

fbdescamps, 2017-01-12:
I agree that this information can be deleted. Other opinions?