snoplus/orca #29: Delete detector wide buttons from XL3 GUI

tlatorre-uchicago, 2016-02-01:
At the bottom of the XL3 GUI there are three detector wide buttons:

- panic down
- triggers on
- triggers off

Although the panic down is not as confusing, the triggers are very easy to mistake for a local change to that specific crate.

fbdescamps, 2016-03-08:
This should be partly replaced by the global always-on-top detector-wide box, which will have the 'emergency' buttons: 

 * big red 'PANIC DOWN'
 * big yellow/orange 'TRIGGER OFF'

The detector-wide triggers on is in the danger zone of the SNO+ control panel run control.

tlatorre-uchicago, 2016-05-04:
@fbdescamps , Now that we have these buttons in the SNO+ GUI, what do you think about removing them from the XL3 GUI?

fbdescamps, 2016-05-05:
Hmm.. so, I think we can delete them from the XL3 GUI, if we are absolutely sure that it works from the SNO+ GUI.

tlatorre-uchicago, 2016-05-23:
Fixed in #123.

closed 2016-05-23 by tlatorre-uchicago