snoplus/orca #38: EsumHi and EsumLo labels are swapped in the MTC Gui

jcaravaca, 2016-02-11:


tlatorre-uchicago, 2016-02-16:
Where are they switched? Just in the GT mask?

jcaravaca, 2016-02-16:
The actual labels in the GUI are switched according to the definition in the mtcDataBase.
I.e. EsumLo should be on the top and EsumHi right after this.

tlatorre-uchicago, 2016-04-13:
I'm still confused about this. What do you mean according to the definition in the mtcDataBase? Can you point me to some source code. According to the trigger document page 70, these are correct.

jcaravaca, 2016-04-13:
In mtcDataBase the ESumLow is right before the ESumHi:
https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO%2B/MTC/ORMTCModel.m#L143
But in the GUI the labels themselves are swapped: the ESumHi is before the ESumLow. As the mtcDataBase is filled following the order of that table, there is no consistency. If that makes sense now I can correct it in 2 secs.

tlatorre-uchicago, 2016-04-13:
So, you are talking about the MTCA thresholds? I thought you were talking about the GT mask.

jcaravaca, 2016-04-13:
Yeah, I meant the threshold values under the 'settings' tab.

tlatorre-uchicago, 2016-04-13:
Ah OK. Yeah, please submit a pull request!

tlatorre-uchicago, 2016-04-13:
Fixed by #58.

closed 2016-04-13 by tlatorre-uchicago