snoplus/orca #40: NSLog blocks

marzece, 2016-02-16:
Maybe everyone already knew this but as far as I can tell [this](https://github.com/snoplus/orca/blob/master/Source/Main%20Dialogs/StatusWindow/ORStatusController.m#L194) chunk of code gets called each time NSLog gets called.
The waitUntilDone:YES means that function will block until the Main Thread is not busy. So any thread that calls NSLog will sit there doing nothing until the Main Thread is free. But it's certainly possible that the Main Thread is waiting for that other (now blocking) thread to do something. So both threads will block basically for eternity.

I discovered this problem by that exact thing happening, except instead of eternity it was 2 seconds (there was a timeout). 

There are two solutions as I see it. 
1.) Change the waitUntilDone option from YES to NO. Then it won't block. But changing how NSLog works will effect pretty much everything in ORCA. 
2.) Make sure NSLog statements in critical threads (such as the ones that receive commands from the XL3) are all launched in yet another separate thread

I'm in favor of talking to Mark Howe and depending on what he says going for option 1. But I thought I'd bring it up here to see if anyone else has thoughts on the matter.

marzece, 2016-02-16:
I may have spoken too soon. The (oneway void) return type is apparently supposed to keep the function call from blocking, [link](http://stackoverflow.com/questions/5494981/use-case-of-oneway-void-in-objective-c). Nevertheless the call does block and changing the waitUntilDone from YES to NO does seem to fix it. So the oneway void thing just makes things slightly more confusing.

marzece, 2016-02-18:
I sent Mark Howe an email regarding this. He said

> Please do not change any objects other than SNO objects. The NSLog code is done that way for thread safety. If you are running into a deadlock, you can implement a logging method in your thread that prints via the main thread.

tlatorre-uchicago, 2016-03-24:
Fixed

closed 2016-03-24 by tlatorre-uchicago