snoplus/orca #51: MTCA  masks needed to be unchecked then rechecked to prevent an error

marzece, 2016-04-07:
mtcatLoadCrateMask: ERR crate 0 is not plugged into mtca 4
mtcatLoadCrateMask: ERR crate 0 is not plugged into mtca 5
mtcatLoadCrateMask: ERR crate 0 is not plugged into mtca 6

will pop up

The problem is solved by toggling the checkboxes for the OWLELO, OWLEHI, and OWLN relays. It must be some stupid initialization problem which gets fixed whenever a checkbox is changed. Need to fix this!

tlatorre-uchicago, 2016-05-04:
Discovered the problem here: https://github.com/snoplus/orca/blob/830ef12917d8646d0299457ae622c8b1a2d8416a/Source/Objects/Custom%20Hardware/SNO%2B/MTC/ORMTCModel.m#L836.

Looks like somebody copy pasted several lines and forgot to update them.

tlatorre-uchicago, 2016-05-04:
Fixed by #74.

closed 2016-05-04 by tlatorre-uchicago