snoplus/orca #68: Flipping HV on/off state in SNO+ control GUI

fbdescamps, 2016-05-04:
When starting ORCA and  turning on HV on a single crate, we noticed that the ```PMTs HV is ON/OFF``` button was flipping (on off on off on off, etc).

jcaravaca, 2016-05-04:
Is that the global control button? Was only that or the PANIC DOWN being grayed out as well?

fbdescamps, 2016-05-04:
Both were in flipping state, I think it was like this:
 * HV ON was combined with enabled panic down
 * HV OFF was combined with disabled panic down

But the HV was ON.

fbdescamps, 2016-05-04:
[image]

jcaravaca, 2016-05-04:
Does the HVMaster panel look like normal? Indicates ON/OFF the correct crates?

fbdescamps, 2016-05-04:
Ah, we did not check this. Also, for some unknown reason, the flipping stopped and it went to ```HV ON`` when it indeed was.

tlatorre-uchicago, 2016-05-04:
It's flipping again, and the HV master tab is stable with just crate 6
showing HV ON.

BenLand100, 2016-05-09:
https://github.com/snoplus/orca/blob/master/Source/Experiments/SNOP/SNOPController.m#L460
In the event that there is a value in ```aNote``` the ```globalHVON``` variable is not set as an OR of XL3 states. This notification is shipped with a value for ```aNote``` for each connected XL3 so ```globalHVON``` should be set by looping over the XL3s outside that ```if``` statement and all will be good.

closed 2016-05-16 by jcaravaca