snoplus/orca #69: It takes 60 seconds for the HV thread to re-appear after an ML403 reset button press

fbdescamps, 2016-05-04:
That's pretty long...

BenLand100, 2016-05-06:
Some background: the HV monitoring thread is waiting on the readHvStatus method to time out after the XL3 disconnects. This evidently takes a long time. After the read times out the monitoring thread exits as expected, but the timeout triggers Orca to think the XL3 disconnected again. After it reconnects a second time everything is fine again. Tony and I think this will take some significant modifications to how XL3 responses are handled in Orca to fix, but besides the excessive wait everything works "as expected".

tlatorre-uchicago, 2016-05-07:
This issue however made it *very* confusing for us in a panic situation on shift yesterday. We were trying to debug why two XL3s wouldn't connect properly to ORCA (they were disconnecting and then reconnecting repeatedly), and there was no way for us to tell from the logs whether a disconnect was caused by the XL3 or by this HV thread.

Based on this experience, I would actually classify this as a pretty serious bug, since it will *really* confuse an operator (including an expert operator) during an emergency situation.

I think the timeout length is not really the issue, the fundamental solution needs to be that after a disconnect, the HV thread is killed.

BenLand100, 2016-05-07:
As it is currently the HV thread is marked to exit when the XL3 disconnects, but it is caught blocking on that read that can't finish until it times out. This leads to the undesirable condition: the blocking call eventually times out and makes orca think the XL3 has disconnected again. I still think that when the XL3 disconnects the pending waiting commands should be cleared/destroyed/immediately-timed-out/whatever instead of hanging around and causing problems. To really drive this point home, even if we killed the HV thread immediately on disconnect (technically unsafe) I believe the command would _still_ be waiting in the queue, time out, and cause the exact same problem. I.E. it's technically not the HV thread causing the second disconnect, just the fact that other threads can send packets to the XL3 and that is being mishandled by the XL3 packet queuing code in Orca.

BenLand100, 2016-05-07:
Turns out making the HV thread exit on disconnect and making the XL3 packet handling code disconnect-aware may be degenerate. Essentially this loop https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO%2B/XL3/XL3_Link.m#L584 needs to exit on a disconnect (and remove its command from the queue) with some failed status. That should solve this issue.

tlatorre-uchicago, 2016-05-07:
Yes, I think that plus all the other functions need to be exited upon a disconnect.

The queue is a queue of *incoming* packets, so killing the HV thread would indeed fix the problem. The queue should be cleared, but the timeout happens because it doesn't find the packet it's looking for, not because there is a packet in the queue.

Also, the timeout from that call *actually* causes a disconnect between ORCA and the XL3 server, it's not just a log message.

BenLand100, 2016-05-07:
Ah one thing I forgot to mention on the call was that force killing the HV thread while it is waiting in this method is particularly dangerous because it could be killed while holding a lock which would forever deadlock XL3 communication.

tlatorre-uchicago, 2016-05-07:
Is it possible to have a some sort of lock inside the XL3 Link that determines who is using it at a given time and after a disconnect it waits until it secures the lock to reconnect.

That would solve the issue of reconnect - disconnect. The 60 seconds may still be too long in which case we can figure out another solution, but I think that solves the main problem.

Not sure if this is the best way to attack this problem.

BenLand100, 2016-05-07:
So basically counting the threads using the XL3 connection? Sure, we could have a locked integer count initialized to zero in the XL3 link that is locked incremented and unlocked when the readXL3Packet is entered and is locked decremented and released before readXL3Packet exits and only allow reconnects when this count is zero. That's actually a pretty graceful way to handle it, but we'd still have to wait the full timeout on each disconnect as you say.

Could additionally have another flag in the XL3 link that is set to 'disconnected' on a disconnect which is checked by readXL3Packet to cause an early abort in the event of a known disconnect. This flag would be set back to 'connected' after the thread count goes to zero and the XL3 reconnects. This flag already exists, I think, it's just a matter of making sure the reconnect only happens after all threads are done and also checking it in readXL3Packet. 

I can implement this tomorrow unless you feel like you'll get to it before then?

tlatorre-uchicago, 2016-05-07:
If you could solve this by tomorrow that would be awesome. Somehow we have to make sure that another thread can't call into that function after the reconnect has started, but not completed though.

tlatorre-uchicago, 2016-05-08:
Fixed by #92.

closed 2016-05-08 by tlatorre-uchicago