snoplus/orca #73: SNO+ GUI became unresponsive

tlatorre-uchicago, 2016-05-04:
At some point during running today, the SNO+ GUI became completely unresponsive, and we had to restart it.

BenLand100, 2016-05-07:
I noticed this happening with the XL3 gui when the gui was modified from another thread. If the run start threads are making gui changes, these will need to be dispatched to the main thread.

jcaravaca, 2016-05-10:
Fixed in #75

jcaravaca, 2016-05-10:
Fixed in #75

tlatorre-uchicago, 2016-05-10:
We had to force quit Orca during our shift because of an unresponsive SNO+
GUI, so I don't think this is fixed.

fbdescamps, 2016-08-18:
What is the status? Has this been confirmed to still be an issue? @jcaravaca

jcaravaca, 2016-08-18:
It should be fixed. The bug didn't show up during the last days of running, so feel free to close it and we can reopen it if the problem is back.

fbdescamps, 2016-10-20:
I am closing this issue for now, see @jcaravaca last comment.

jcaravaca, 2016-12-01:
The GUI changes are not dispatched to the main thread anymore following a comment of @tlatorre-uchicago in #75, but it seems to be lost. Could you remind me what was your worry? It seems that doing that is standard practice: http://stackoverflow.com/questions/15813764/posting-nsnotification-on-the-main-thread

tlatorre-uchicago, 2016-12-01:
My comment there was just that dispatching to the main thread doesn't belong in the setter, especially if it's asynchronous.

If a member variable needs to be set from a thread and the main thread also accesses it, then you need to use locks or some other method of synchronization.

jcaravaca, 2016-12-08:
Mark H. seems to have seen that problem very rarely in the main Run Control as well. Attached comment from Mark H. for the records:

> I have seen that problem (rarely) also. Never been able to catch it with the debugger, tho. If you can help solve the problem I would be grateful. Whatever the cause, it seems to be just the Run Control dialog -- I have never seen it elsewhere. I also suspected that the GUI was being updated from some other thread. That may be it, but I have not been able to track it down so far. A global 'freeze' of that dialog could also mean that all of its notifications got 'de-registered' somehow. Don't know how that could happen tho.

tlatorre-uchicago, 2016-12-09:
I saw this happen today on both of them at the same time. This suggests it might be worth looking for a notification in the RunModel which gets posted on the wrong thread.

fbdescamps, 2016-12-30:
I can add that I have seen this happen a few times now. Until this is resolved, we should make it an hourly check to make sure the SNO+ main GUI is still responsive. @marzece can this be added?

fbdescamps, 2017-01-16:
To follow up, it seems that there was an issue to close the GUI in this shift: http://snopl.us/shift/view/e2083a8fbcf14054baeb0c07431491f4 

Quote:
```14:58 Sno+ Control Panel froze. We were not able to close the window until 15:00. Opened a new window, and everything is working again```

icoulter, 2017-01-23:
So we had a freeze on this evenings shift. We were restarting runs for Freija to testing the nearline and on pressing RESYNC, things froze and the time elapsed stopped counting and we couldn't switch tabs. However pressing RESYNC and STOP still worked. Unfortunately, I didn't try any other buttons before closing the window.
I don't know if this might give some clue as to why this is happening?

klabe, 2017-01-29:
This happened on our shift this evening.  The GUI stopped updating right when it was going to roll over into the next run -- perhaps that's a hint?  We were able to get the GUI to update with the correct run number by clicking on the "Detector State" tab, but it was still frozen after that.  Closing and reopening the window seemed to fix the problem.