snoplus/orca #79: failed to connect outlet on bootup

tlatorre-uchicago, 2016-05-05:
The following message is printed to stderr when you first start ORCA:

```
2016-05-04 20:37:46.089 Orca[15244:10674893] Failed to connect (problemMatrix) outlet from (ORFec32Controller) to (NSMatrix): missing setter or instance variable
2016-05-04 20:37:51.963 Orca[15244:10674893] Failed to connect (xilinxFileField) outlet from (ORMTCController) to (NSTextField): missing setter or instance variable
```

BenLand100, 2016-05-06:
Assuming no one is currently modifying these guis, the fix is to click the little "x" on the offending outlet in the outlets tab (right arrow in a circle) with the correct GUI element selected.

For the ORFec32Controller the offending element is the check box grid on the DB tab, and for ORMTCController it's the Label next to the Xilinx FIle... button on the settings tab.

tlatorre-uchicago, 2016-05-06:
Damn. OK. I will look into this.

marzece, 2016-05-07:
Maybe while we're at it should just delete the DB checkboxes . They're obviously unused and I think the plan is to implement a channel db outside of orca anyways, right?

pershint, 2016-05-17:
Take a look at pull request #121 for a fix.  A note that the DB checkboxes matrix is still not being used anywhere; it's just linked to a defined matrix until we would want to implement it in the controller.

tlatorre-uchicago, 2016-05-23:
Fixed in #121.

closed 2016-05-23 by tlatorre-uchicago