snoplus/orca #84: normal mode on run start

klabe, 2016-05-06:
Crate 0 did not enter normal mode on run start.

marzece, 2016-05-06:
We saw something like this happen with crates 9 & 10 earlier. Basically when we turned the crates on and initted them most crates were in normal mode after that. However that was not the case for 9 and 10. 
Here's the shift report (though I've already given more info here than I did in the report) 
http://snopl.us/shift/view/30aca1acdff6411fa9c8bd6225c51fff

fbdescamps, 2016-12-30:
Is there any update on this one? @tlatorre-uchicago ?

tlatorre-uchicago, 2017-01-10:
I haven't been able to reproduce this. I suspect it's a non issue. I don't know exactly what caused the earlier problems but here are my suspicions:

1. If the XL3 window drop down selection was set to NORMAL mode but never actually sent, that would cause ORCA to think that the crate was in NORMAL mode (I think?) which means that it wouldn't try to set it to normal mode on run start.

2. The issue Eric reported seems to be unrelated to run starts. The XL3 is not put into normal mode by default after a crate init, so that is expected.

I'm going to leave this issue up though just as a reminder that it might be a good idea to set the crate to normal mode after loading the settings and not at run starts, since I think that is when most people naturally expect it to happen.

tlatorre-uchicago, 2017-01-19:
Pull request #254 sets the XL3s to normal mode after loading the hardware settings.

closed 2017-01-22 by tlatorre-uchicago