snoplus/orca #90: When you stop a run pedestals continue to be emitted. Pulse GT's however stop.

marzece, 2016-05-07:
The intended behavior is probably to stop both pedestals and  pulse gts.

tlatorre-uchicago, 2016-12-17:
When you stop a run, the GT mask is now cleared so there are no triggers. I don't think that the pulser is explicitly stopped, so it may still send out pedestal pulses if pedestals were enabled. Do you think we should stop the pulser on a run stop?

marzece, 2016-12-17:
I think so. Even if there's no obvious consequences I'm sure someday someone will be surprised to find out the MTC is emitting pedestals when they didn't think that it was. Basically having it emit pedestals despite everything else being stopped violates the [principle of least astonishment](https://en.wikipedia.org/wiki/Principle_of_least_astonishment).

tlatorre-uchicago, 2016-12-17:
Fixed in #213.

closed 2016-12-17 by tlatorre-uchicago