snoplus/orca #94: Orca should check XL3 is in normal mode before init-ing w/ Xilinx

marzece, 2016-05-08:
I'm not sure if orca actually knows if an XL3 is normal mode or not for sure. But if it does it should do this check.

pershint, 2016-05-17:
Just to check Eric, should the XL3 be in "init mode" before init-ing with the Xilinx?

marzece, 2016-05-17:
Yes, due to some issues that popped up during the workshop it was decided that one should not do an init w/ xilinx unless an xl3 is in init mode. In most cases this won't be a problem since xl3s are in init mode after a power cycle. But if someone hits init w/ xilinx on an xl3 that's been running and is already initt-ed then orca should switch it to init mode before moving forward with the init.

pershint, 2016-05-20:
Fixed in pull request #125.

closed 2016-07-18 by tlatorre-uchicago