snoplus/orca #101: charge injection freezes Orca

ecaden, 2016-05-09:
Starting and stopping charge injection takes a long time (~15s) and Orca hangs with the beachball when this is going on. It's essentially frozen.

tlatorre-uchicago, 2016-05-09:
Is this reproducible or does it only happen sometimes?

ecaden, 2016-05-09:
It happened every time the charge injection state was changing.

BenLand100, 2016-05-13:
Which button is this on the GUI? If Orca hangs it means that action is being done on the main thread when it should be dispatched to another thread.

tlatorre-uchicago, 2017-06-01:
The code responsible for being so slow is here: https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO%2B/XL3/ORXL3Model.m#L3251.

The reason is that it's doing all the low level reads and writes on the FEC. There exists an XL3 packet already for setting all of this up at once, so this would significantly reduce the amount of time it takes to set up charge injection.

See https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO%2B/XL3/PacketTypes.h#L546 for the packet arguments to set up charge injection with a single command.