snoplus/orca #104: unexpected dialog box when opening HV tab

tlatorre-uchicago, 2016-05-10:
Simply clicked the HV tab for crate 4 which was at HV and this dialog box popped up.
[image]

BenLand100, 2016-05-13:
We'll need to think about what to do in the case that the target in the .orca file is lower than the readback value on init. This is what triggered that popup, though it's a little mysterious that it happened immediately on opening. Perhaps the target box took focus and on losing focus the range checks were run?

tlatorre-uchicago, 2016-05-23:
Is this still an issue? ORCA now doesn't send any HV setpoints at start up now right?

I don't think it could have been the target box focus as this happened *immediately* after hitting the HV tab at the top.

Also, are the OK and the message swapped?

BenLand100, 2016-05-23:
I'm not sure how it happened still, so that's outstanding. That's correct that orca does not send setpoints to the XL3 on start up, but it does update its internal setpoint to the readback on starting. So what I think happened was this setpoint (while below nominal) was above the target saved in the .orca file and by some unknown cocoa magic this check was run when the tab was rendered.

Yeah the "OK" and message are swapped - same for the the waring if you try to press HV Off when the supply is ramped up. It's on my list but I haven't touched that code yet so hasn't yet been fixed. Presumably it's just arguments that need to be swapped.

fbdescamps, 2017-01-05:
@BenLand100 Any update on this issue?

fbdescamps, 2017-01-06:
Yes, I agree that we want this. If there are no objections, I'll be closing this issue. Thanks!

closed 2017-01-06 by fbdescamps