snoplus/orca #124: Orca thinks HV is on after crate init

pershint, 2016-05-20:
I was doing some ORCA testing on the teststand and noticed after performing a crate init with xilinx while at HV, ORCA thought the crate's HV was on when it was not.  Is this expected/has it been seen before?

Steps to replicate:
 - Init crate with xilinx at startup
 - Turned on HV
 - Ramped up to 500 V
 - Hit "init xilinx, safe" to re-init crate
 - Voltage dropped to zero and high voltage is off (verified by HV LED not being on)
 - Press "accept readback"
 - ORCA thinks high voltage is still on ("HV ON" button grayed out, "HV OFF" is pressable)

BenLand100, 2016-05-23:
I'm not exactly sure what the crate init does. This would be expected if e.g. the init sets the setpoints to 0 but does not change the supply state. Alternatively orca may not be reading back the button state after "accept readback" is pressed. I'll look into this aspect.

BenLand100, 2016-05-23:
@pershint so the readback accepting does not check the button state (as this shouldn't have changed under a normal readback discrepancy), but since initializing the crate changes the high voltage status outside of orca you should call safeHVInit https://github.com/snoplus/orca/blob/fe8ee334db930776788474f99c6f7682ab1da641/Source/Objects/Custom%20Hardware/SNO%2B/XL3/ORXL3Model.m#L3360 on the appropriate XL3 to update the high voltage control. This is essentially what the panic down does now.

tlatorre-uchicago, 2016-12-17:
I think this is still an issue, but it shouldn't be possible now from ORCA to do a crate reset (equivalent of a crate init with Xilinx before) while at HV, so I'm going to close the issue.

closed 2016-12-17 by tlatorre-uchicago