snoplus/orca #127: Got HV discrepancy for crate 19 supply B

tlatorre-uchicago, 2016-06-06:
When starting ORCA on the teststand, I got an alarm that said HV discrepancy for crate 19 supply B, but there is no supply B for crate 19.

BenLand100, 2016-06-07:
The XL3 must have reported that the switch for the B supply was enabled (possible even if there is no supply B for that crate) as the code that fires the alarm only runs if the switch is on according to the XL3 model. The garbage readbacks likely then triggered the alarm. I don't suppose you looked at the A B lights on the front of the XL3 when this happened?

I could hardcode supply B to not exist for all but crate 16 in the alarm code as is done elsewhere in the XL3 code which would at least suppress completely bogus alarms... unless someone adds a supply to a crate.

tlatorre-uchicago, 2016-06-07:
Unfortunately, I wasn't on site, so I don't know what happened with the LEDs. I'll try and contact someone on site if it happens again.

It shouldn't report that the supply is enabled though right? I mean we now write to the ADC test register on start up which allows us to read back the HV and this should also give us the correct HV switch readings? @bonventre

fbdescamps, 2016-08-18:
Is there an update or resolution for this issue? @BenLand100

BenLand100, 2016-08-18:
I still suspect it was an issue with the XL3 at the time - we haven't seen this happen again. Otherwise cosmic rays are a possibility (this couldn't happen otherwise without inaccurate information reported by the XL3).

bonventre, 2016-08-18:
The hv enable is just a bit in a register that gets reset on bootup / initialization, so no real reason for any issues from the xl3

tlatorre-uchicago, 2016-12-17:
Fixed in #203.

closed 2016-12-17 by tlatorre-uchicago