snoplus/orca #128: Crashes occured during run starts and save operation

marzece, 2016-06-07:
I suspect the issue(s) that we saw were primarily caused by the fact that we had dragged a new FEC from the catalog into crate 18 (which wasn't on). Since it never got values from an ECAL document or a crate init or anything like that the card had a number of un-initialized values that perhaps were just random chunks of memory. So when we tried to do a run start/save the .orca file it tried to write those values to disk. And that caused a seg-fault.  Perhaps related is that while trying to debug that issue I had set a break point and I noticed that at the breakpoint the snopmodel's pointer would flip quickly between NULL and a reasonable value. I suspect that is caused by a thread that wasn't affected by the breakpoint alloc-ing and dealloc-ing the snopmodel.

tlatorre-uchicago, 2016-06-07:
Do you have the traceback?

marzece, 2016-06-07:
I think you already noticed this but for the sake of posterity the logs are in [this shift report](http://snopl.us/shift/view/9cb0e4228c93467d8ebe0c0e43d6a69a)

marzece, 2016-07-19:
Fixed (kinda) in #134

closed 2016-07-19 by marzece