snoplus/orca #130: CMOS polling output should be unsigned

marzece, 2016-06-30:
Currently you can get output like the following when a very high rate shows up. 
Why you might get rates that high is a whole other question. 

ch24-31:      298       298       298       298       298       298       298       299 
slot: 13
ch00-07:      299       299       299       299       299       299       299       299 
ch08-15:      299 -2147483648       299 -2147483648       299 -2147483648       299 -2147483648 
ch16-23:      299 -2147483648       299 -2147483648       299 -2147483648       299 -2147483648 
ch24-31:      299 -2147483648       299 -2147483648       299 -2147483648       299 -2147483648

tlatorre-uchicago, 2016-06-30:
I suspect this is because it doesn't handle the CMOS counter rolling over correctly.

fbdescamps, 2017-01-12:
@marzece What is the status here?

marzece, 2017-01-13:
I haven't touched this since my last attempt failed. Since CMOS polling might soon be moved out of orca this may become a non-issue.

fbdescamps, 2017-02-09:
I am leaving this open until we have the ORCA-independent CMOS polling.

tlatorre-uchicago, 2017-07-14:
I wonder if the reason your pull request didn't work is because the counters are actually 31 bits instead of 32 bits. I was trying to figure out the size of the register and came across this: https://github.com/pennneutrinos/fullml403/blob/8cb8b8336b8ba474485000e85043ff35882a2970/src/FECUtils.c#L250 which suggests that it's only 31 bits.

@marzece Am I reading that correctly? I couldn't find any other docs which explicitly said how many bits there are in the count register.

marzece, 2017-07-15:
Huh, whaddya know. From the ML403 code for getting cmos counts
```
readResults = doRW(0x0, CMOS_INTERN_TOTAL(j) + FEC_SEL*i + READ_REG);
errorBit = (readResults & 0x80000000) ? 1 : 0;
...
readResults &= 0x7FFFFFFF;
results->count[numSlots*32+j] = readResults;
```
It would seem it was only 31 bits. That very likely did cause a lot of the issues I ran into with this.