snoplus/orca #131: NSUndoManager crash caused by XL3 polling/monitoring code

marzece, 2016-06-30:
Here's a stack trace from when the error occurred. Details viewable here 

http://snopl.us/shift/view/fe9bf82f357b44a38d246385ffe7d965

2016-06-29 20:57:48.854 Orca[43669:10592953] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'forwardInvocation:: NSUndoManager 0x28946ee0 received forwarded invocation while invocation target is nil. Call prepareWithInvocationTarget: before invoking setIsPollingXl3:'
*** Call stack at first throw:
(
	0   CoreFoundation                      0x9127b343 __raiseError + 195
	1   libobjc.A.dylib                     0x92d43a2a objc_exception_throw + 276
	2   CoreFoundation                      0x9127b25d +[NSException raise:format:] + 141
	3   Foundation                          0x90aee124 -[NSUndoManager _forwardTargetInvocation:] + 169
	4   Foundation                          0x90aee069 -[NSUndoManagerProxy forwardInvocation:] + 43
	5   CoreFoundation                      0x911b7122 ___forwarding___ + 482
	6   CoreFoundation                      0x911b6f1e _CF_forwarding_prep_0 + 14
	7   Orca                                0x005303bb -[ORXL3Model setIsPollingXl3:] + 90
	8   Orca                                0x0053d7d3 -[ORXL3Model readHVStatus] + 301
	9   Orca                                0x00543bcc -[ORXL3Model(private) _hvXl3] + 2571
	10  Foundation                          0x90ab2dcf -[NSThread main] + 45
	11  Foundation                          0x90ab2d21 __NSThread__main__ + 1538
	12  libsystem_pthread.dylib             0x915e3ecf _pthread_body + 138
	13  libsystem_pthread.dylib             0x915e3e45 _pthread_body + 0
	14  libsystem_pthread.dylib             0x915e1f0e thread_start + 34
)
The issue is almost certainly here. 
https://github.com/snoplus/orca/blob/fe8ee334db930776788474f99c6f7682ab1da641/Source/Objects/Custom%20Hardware/SNO%2B/XL3/ORXL3Model.m#L580

tlatorre-uchicago, 2016-07-12:
I'll try to take a look at this soon.

tlatorre-uchicago, 2016-07-18:
Yep, so the problem is that the readHVStatus method is running in a separate thread and is not calling setIsPollingXL3 on the main thread here: https://github.com/snoplus/orca/blob/fe8ee334db930776788474f99c6f7682ab1da641/Source/Objects/Custom%20Hardware/SNO%2B/XL3/ORXL3Model.m#L3431.

tlatorre-uchicago, 2016-07-18:
Fixed in pull request #135.

closed 2016-07-19 by marzece