snoplus/orca #132: DB Url request sometimes fails mysteriously in SNOPController

tannerbk, 2016-07-11:
This is Eric not Tanner, he left himself logged into github on operug.

   The issue occurs around line [1950 of  SNOPController.m](https://github.com/snoplus/orca/blob/master/Source/Experiments/SNOP/SNOPController.m#L1951).
Basically sometimes the NSError variable is set by the call to sendSynchronousRequest and sometimes it's not set. When an error does occur the output simply says the format of the data is not valid.

Below is a (slightly edited) version of the code. I added extra debug statements. The output of which is also below.
  ```
    NSLog(@"self = %@\n",self);
    //Now query DB and fetch the SRs
    NSString* urlString = [NSString stringWithFormat:@"http://%@:%@@%@:%u/orca/_design/standardRuns/_view/getStandardRuns",[model orcaDBUserName],[model orcaDBPassword],[model orcaDBIPAddress],[model orcaDBPort]];
    NSString* link = [urlString stringByAddingPercentEscapesUsingEncoding:NSUTF8StringEncoding];
    NSLog(@"link : %@\n",link);
    NSURLRequest* request = [NSURLRequest requestWithURL:[NSURL URLWithString:link] cachePolicy:0/*NSURLRequestReloadIgnoringCacheData*/ timeoutInterval:1];
    NSURLResponse* response=nil;
    NSError* error=nil;
    NSData* data= [NSURLConnection sendSynchronousRequest:request returningResponse:&response error:&error];
    NSString *ret = [[[NSString alloc] initWithData:data encoding:NSUTF8StringEncoding] autorelease];
    
    NSDictionary *standardRunTypes = [NSJSONSerialization JSONObjectWithData:[ret dataUsingEncoding:NSUTF8StringEncoding] options:0 error:&error];
    
    if(error) {
        NSLogColor([NSColor redColor], @"here\n errror: %@\n data: %@\n",error,data);
        [model setStandardRunType:@""];
        return;
    }
```
    
 self = <SNOPController: 0x2a432c00>
16-07-11 12:54:08 link : http://USER_REDACTED:PASSWORD_REDACTED@couch.snopl.us:80/orca/_design/standardRuns/_view/getStandardRuns
16-07-11 12:54:09 here
 errror: Error Domain=NSCocoaErrorDomain Code=3840 "The data couldn’t be read because it isn’t in the correct format." (No value.) UserInfo=0x35603550 {NSDebugDescription=No value.}
 data: (null)

However if you take the link and fill in the appropriate user name and password ( which I edited out) the results look perfectly serviceable and I can't think of a reason why it would sometimes not be of valid format and it sometimes would be. 

One potential explanation/solution would be to bump the timeout time. Though 2 seconds seems like it should be plently long but maybe not, I don't know.

tlatorre-uchicago, 2016-07-18:
Is it repeatable at all? I mean if you get the error and immediately run the same function does it produce the same error?

marzece, 2016-07-18:
Sometimes. I would kill and recompile orca (without changing anything) and sometimes get the error again and sometimes not. I think re-compiling orca isn't necessary though. I think opening and closing the snop-model also calls the code where the error can occur, though I don't know if it doing that will re-roll whatever (seemingly) random condition causes this error.

tlatorre-uchicago, 2016-12-17:
Haven't seen this in a while, so I assume it's fixed.

closed 2016-12-17 by tlatorre-uchicago