snoplus/orca #139: HV controls: grey out HV controls when in physics run.

fbdescamps, 2016-08-11:
As discussed in the DOW2016: we might want to prevent the operator from changing the HV (other than panic down) while in a physics run. Should be allowed in maintenance run.

BenLand100, 2016-08-18:
To clarify: changing HV should only be allowed in maintenance runs or changing HV should only be disallowed in physics runs?

fbdescamps, 2016-08-18:
My opinion is that we should only allow changed to HV in maintenance runs. We will off course allow individual crate panic downs. Other opinions/concerns anybody?

tlatorre-uchicago, 2016-08-18:
I think we shouldn't overcomplicate the controls. Since there is already a plan to only allow changes with an expert password, we shouldn't add in more checks that require cross model communication.

fbdescamps, 2016-11-29:
@tlatorre-uchicago @BenLand100 What is the decision on this? Please retire and replace with a better formulated issue if required.

BenLand100, 2016-11-29:
@jcaravaca implemented non-expert locking in such a way as to handle this. My understanding is that unlocking puts you in maintenance mode, and otherwise the controls are locked. Unless I am mistaken this can be closed. (But otherwise modifications should be done to existing locking logic as tony suggests.)

jcaravaca, 2016-11-30:
The new operator mode PR fixes this: the HV will be available only in expert mode if the run is stopped or in maintenance. I'd keep this up and include it as a HV test once Ben's and my PRs are merged.

tlatorre-uchicago, 2016-12-17:
@jcaravaca Do you still want to leave this up?

jcaravaca, 2016-12-17:
Do we want this to happen? If there is an emergency in the middle of a Physics run we might want to ramp down a crate. The XL3s HV controls are blocked but that's not true for the SNO+ GUI.

fbdescamps, 2017-01-04:
This might be related to issue #221 , where it is suggested that we add 'ramp down' to the HV master. If we want to allow operators to ramp down a crate we need to either:
 * allow access to HV controls
 * add ramp down to HV master

jcaravaca, 2017-01-08:
I would go for allowing access to the HV controls. This is not an emergency operation unlike Panic Down so it's reasonable to use the XL3 panel. The other question is that whether we really want to allow the operator to perform this task and in which situation that'd be useful.

tlatorre-uchicago, 2017-01-19:
I think the current plan is to enable ramp down and HV off for non experts.

closed 2017-01-19 by tlatorre-uchicago