snoplus/orca #140: HV Relays Disabled Count

BenLand100, 2016-08-12:
Add count of disabled relays to main GUI for shift report

fbdescamps, 2017-03-03:
Just chatting with @ecaden , we think a count of the total number of disabled relays would be nice. @BenLand100 , would you have time to implement this? Where would we display it?

icoulter, 2017-03-03:
I quite like the current system where you have to actually check the HV tabs as it forces you to know which crates and slots are off.

BenLand100, 2017-03-03:
I've heard arguments similar to Ian's whenever this has come up. It's not a hard thing to count and could be added to anywhere on the GUI (when I get around to it) but it sort of defeats the purpose of the shift start check.

ecaden, 2017-03-04:
I can understand both sides of the argument. I was tidying up the operator manual, removing all referenced to SHaRC, when I saw they had a tally on their gui. We can leave it as a manual count.