snoplus/orca #141: XL3 Graphic to red

parnmatt, 2016-08-17:
Not a huge thing, just would be nice if the XL3 in the ORCA crate looked more like an XL3, i.e., at least red.

Would be easier for operators to quickly identify the XL3.
I know I would find it much easier.

tlatorre-uchicago, 2016-08-17:
If someone sends me a nice high quality photo of an XL3 from the front in good lighting, I will add it to ORCA.

I agree this would be nice.

marzece, 2016-08-17:
I actually have a branch where I did this sitting around. I didn't like the new look of it so I kinda dropped it. But I can add a PR anyways and people can decide if the change is so aesthetically offensive that it will ruin orca

marzece, 2016-08-17:
Here's the image I used if anyone else wants to try.
[image]

tlatorre-uchicago, 2016-08-22:
fixed in #144.

closed 2016-08-22 by tlatorre-uchicago