snoplus/orca #150: HV issue on teststand

tlatorre-uchicago, 2016-08-23:
I was doing some testing on the teststand and some strange behavior happened on the teststand. The teststand was running commit 2e5939cfb346d2a522d75f29ead3f1b32d5dfe60. Here is the sequence of events:

- turn on the crate
- ORCA alarm for HV B read value differs from set point
- restart XL3 server twice
- ORCA alarm for HV A read value differs from setpoint
- press HV on
- readback is ~3000 V, current is 47 mA

[image]
[image]

- not sure what to do at this point because the voltage shouldn't be at 3000 V, and I don't want to press accept readback since that will set the setpoint to 3000 V
- press panic down
- panic down maybe succeeds or maybe fails? XL3 message is not clear
- now the GUI doesn't show anything for the power supply and I can't turn HV on or off, only button not greyed out is accept readback but there is no readback to accept

[image]
[image]

I've attached the ORCA log here: [orca_log_hv_teststand_august_23_2016.txt](https://github.com/snoplus/orca/files/433565/orca_log_hv_teststand_august_23_2016.txt).

tlatorre-uchicago, 2016-08-23:
I suspect some of the weird behavior at the beginning was due to the crate not being initialized. I know we added the clock initialization at XL3 bootup so that the HV could be read back. Is there anything else we missed that could cause this behavior? @bonventre

I looked through the panic down code and it is intended to fail if it can't turn triggers off, so the panic down indeed didn't work, but that was expected.

BenLand100, 2016-08-23:
Possibly related to #127 ?

BenLand100, 2016-08-23:
The state you ended up in seems to have been because the XL3 disconnected during a setpoint discrepancy. I'll need to add a case to additionally gray out the accept readback button.

The readback of 3kV is consistent with the ADC value having all bits set which is probably a value initilization issue I would think...

bonventre, 2016-08-25:
The ADCs are also enabled on bootup, you can check by reading to address 0x02000007, it should be FFFFFFFF. Besides that not sure - was the HV enabled at the time? On bootup the reset should also clear the hv enable bits etc as well as the dac and setpoints.

tlatorre-uchicago, 2016-08-25:
Yes, the HV was on.

bonventre, 2016-08-29:
So do we think that Orca accidentally set a bad value to the setpoint when you hit accept readback? There is no real way for the xl3 to turn HV on by itself

tlatorre-uchicago, 2016-08-29:
I never hit accept readback. I did turn the HV on though, that's when I started seeing the high readback values even though I had never set a setpoint or ramped up.

bonventre, 2016-08-29:
Ok i see - so the xl3 was never initialized though before this started? Can you try the same procedure again, ignoring the xl3 server stuff? When you hit HV on in orca is it required that you already know that the HV is actually off? Orca could zero out the setpoint before turning the hv on (power on reset should be doing that though)

BenLand100, 2016-08-29:
Orca currently assumes that if the XL3 reports that the HV is off that the XL3 HV setpoint is 0. This is loosely enforced in that Orca won't tell the XL3 to turn the supply off if it thinks the setpoint is too high (Either 50V or 100V - I don't recall off hand). However if some other program (or bug) puts the XL3 in a state where the HV is off but the setpoint is nonzero, Orca could turn the supply on and immediately get a setpoint discrepancy.

bonventre, 2016-08-29:
Sounds good. Is the setpoint assumed to be zero on bootup? We could require a reset or manual setting of the setpoint before assuming it has been zeroed.

BenLand100, 2016-08-29:
It's assumed to be zero on boot yes. I could add code to Orca to zero the setpoint before turning on the HV as you suggest. It might make more sense to have the XL3 do this automatically, but I turning the supply on or off is just a register write not a specific packet that's sent so I'm not sure how hard that would be.

tlatorre-uchicago, 2016-08-29:
@bonventre, the XL3 was not initialized when this happened, but I assumed that since we moved some of the initialization to the boot up that should be ok.

What do you mean ignoring the XL3 server stuff? Do you mean running without the XL3 server? If so, that will be a bit difficult to do since ORCA connects to the XL3 server now instead of listening for a connection from the XL3. I think it's unlikely to be an issue with the XL3 server, but if we eliminate all other possibilities, I can try to build a copy of ORCA to talk directly to the XL3s to test this out.

So you suspect it's that the DAC set point isn't being zeroed on start up? I think that's probably what is happening. We don't however want the XL3 to set the set point to 0 on bootup since that will crash the high voltage if we do an XL3 reset. Is there some register we can use that will survive an XL3 reboot to tell the difference between crate just powered up and XL3 reboot?

@BenLand100 When you say that ORCA won't tell the XL3 to turn the supply off if it thinks the setpoint is too high do you mean the readback voltage or the DAC set point? If it's the DAC set point is that synced up with the actual DAC value set by the XL3?

BenLand100, 2016-08-29:
@tlatorre-uchicago For the software check before switching off supplies it is specifically checking the Orca DAC setpoint with in this case would not necessarily be synced with the XL3 DAC setpoint. Orca's DAC setpoint would either be a value previously sent to the XL3, the last accepted readback, or in the edge case where Orca had just started it would simply be the readback at the moment.

bonventre, 2016-08-29:
The initialization on ml403 boot up should make it so the readback is always working. The XL3 power on reset should clear the setpoint, but otherwise, you always know if the HV is actually off both by the readback and by reading the HV control / status register to tell if the supplies are enabled. If that is off it is safe to write zero to the setpoint.

By ignore the server stuff I just meant don't bother trying to restart it in the same way you did before since I don't think the server has anything to do with the problem.

tlatorre-uchicago, 2016-08-29:
What is the XL3 power on reset?

Also, do I understand correctly that this is the sequence of events that probably happened:

1. Power crate on

At this point the ML403 boot up sets things so that you can monitor the readback voltage, but does not zero the set point, so the set point is essentially random bytes.

2. Turn HV on.

The HV gets turned on and since the set point was already set to something high, the high voltage actually goes to the set point at which point the readback now says that the voltage is high.

Is that correct?

OK, it should be easy to test this out soon on the teststand.

fbdescamps, 2016-11-29:
@tlatorre-uchicago @BenLand100 @bonventre Just a quick reminder of this issue in case we need to test/review this on the teststand this week!

BenLand100, 2016-11-29:
This has been tricky to reproduce, but I think we're satisfied that the issue was likely xl3 initialization which as undergone a major overhaul. On the orca side we will no longer get alarms for supplies that don't exist regardless of what the xl3 reports, and there's no more automatic setting of setpoints. Has anyone encountered a similar issue again?

tlatorre-uchicago, 2016-12-07:
After some debugging on the teststand, the most likely explanation here is a subtle timing issue on the electronics side. @bonventre can probably comment better as to the exact issue.

We have decided upon a two stage solution:

1. Add some code to ORCA to update the HV control status register when the XL3 connects

This will effectively solve the issue.

2. The next time we power cycle the crates we update the ML403 code so that it handles the issue.

bonventre, 2016-12-07:
It is the same issue as always, the ADC needs a signal sent after bootup. We changed the XL3 a while back to send this every time it powers up, but it seems like on occasion this comes too quickly after the power on or the clocks are started. We will just delay the signal in the XL3 code.

Also just to clarify, this signal comes from the test register, not the hv control status.

tlatorre-uchicago, 2016-12-17:
This issue is fixed by #203.

closed 2016-12-17 by tlatorre-uchicago