snoplus/orca #151: SNO+ control panel HV issue

fbdescamps, 2016-08-24:
I noticed, on operag1 (connected to teststand with teststand off), that the SNO+ Control Panel reports that the HV is on. It reports HV on for 16B, see screenshot.
[image]

[image]

Orca log here, looks like the same as for issue #150 
[Orca-log-24082016.txt](https://github.com/snoplus/orca/files/435196/Orca-log-24082016.txt)

BenLand100, 2016-08-24:
@jcaravaca might want to check the logic behind 16B for HV master, though it's odd this hasn't show up before...

BenLand100, 2016-11-28:
It turns out the way the sno+ control panel identifies crate 16B (as index 19) also satisfies a crate 19 (if it exists) which at least explains why the teststand shows up for 16B.

fbdescamps, 2017-02-09:
I believe this is resolved? @BenLand100

BenLand100, 2017-02-09:
Certainly haven't seen this happen for a while. Likely fixed when we sorted out the XL3 init issues.

tlatorre-uchicago, 2017-02-13:
The issue of the XL3 erroneously reporting that the HV on is fixed, but the logic which associates crate 19 with the HV supply for 16B has not been fixed.

jcaravaca, 2017-04-11:
This issue appeared again while testing at the teststand. At this point I think it's understood why it happens and it has an easy fix. However, I think it's useful to leave it like this for HV debugging purpose, since there is no Crate 19 entry in the HV master. I'd close this issue.