snoplus/orca #152: Error in trying to make new version of maintenance run

ecaden, 2016-08-26:
We tried to make new triggers settings (PGT, Pong, Sync) in a new named maintenance run, EricaFreija. This was rejected by ORCA, 

16-08-26 11:24:33 Set GT Mask: 0x02006400
16-08-26 11:25:26 Saving settings for Standard Run: MAINTENANCE - Version: ERICAFREIJA ........ 
16-08-26 11:25:26 MAINTENANCE run saved as standard run. 
16-08-26 11:25:26 Cannot display TEST RUN values. There was some problem with the Standard Run DataBase. 
16-08-26 11:25:26 CouchDB
16-08-26 11:25:26 error : bad_request
16-08-26 11:25:26 reason : invalid_json

:(

fbdescamps, 2016-08-27:
@jcaravaca

fbdescamps, 2016-11-29:
@jcaravaca I suspect that this will be resolved by PR #159?

jcaravaca, 2016-11-30:
It should but feel free to leave this open until we test it at the detector

tlatorre-uchicago, 2016-12-08:
@jcaravaca This is still an issue. I am unable to save to any standard runs.

jcaravaca, 2016-12-08:
@tlatorre-uchicago is that the same error than above? How was the run called? I'll check if it's in the database. Also, did you set couchdb AG or UG?

tlatorre-uchicago, 2016-12-09:
Yes, I saw the same error. I was just trying to save a new maintenance run. The machine on surface where I saw it is running OS X 10.9.5 and ORCA was compiled using XCode version 6.1.1. I tried multiple times and was never able to get it to successfully save a standard run.

jcaravaca, 2016-12-09:
Is this something that happen UG and AG?

tlatorre-uchicago, 2016-12-10:
No, it happens both places. I spent a while today debugging this. It's due to the fact that the function is grabbing all of the settings from the MTC dictionary and one of them happens to be NaN. Here is the json it's posting:

```json
{
	"type": "standard_run",
	"MTC/A,OWLEHi,mV/pC": -10,
	"time_stamp": 1481372273.022906,
	"MTC/D,High10MhzClock": 0,
	"MTC/A,NHit100Med,Threshold": 3000,
	"MTC/A,NHit100Hi,mV/Hit": -36,
	"MTC/A,ESumLow,mV/Adc": 1,
	"MTC/D,PulserPeriod": 20,
	"MTC/A,ESumHi,dcOffset": 4900,
	"MTC/D,Low10MhzClock": NaN,
	"MTC/A,OWLEHi,mV/Adc": 1,
	"run_type": "MAINTENANCE",
	"run_version": "DEFAULT",
	"MTC/A,NHit100Hi,dcOffset": 4900,
	"MTC/A,NHit100Med,m/VAdc": 1,
	"MTC/A,NHit100Lo,mV/Hit": -3.599999904632568,
	"MTC/A,NHit100Lo,dcOffset": 4900,
	"MTC/A,OWLELo,Threshold": 1000,
	"MTC/D,GtCrateMask": 33554431,
	"MTC/A,NHit20LB,mV/Adc": 1,
	"MTC/D,MinDelayOffset": 18,
	"MTC/A,ESumHi,mV/Adc": 1,
	"MTC/D,FineSlope": 0,
	"MTC/A,NHit20LB,dcOffset": 4900,
	"MTC/A,OWLN,dcOffset": 4900,
	"MTC/A,OWLN,mV/Adc": 1,
	"MTC/D,LockOutWidth": 430,
	"MTC/A,NHit20,mV/Hit": -35,
	"MTC/A,NHit20,Threshold": 4045,
	"MTC/A,NHit20LB,mV/Hit": -35,
	"MTC/A,ESumHi,Threshold": 4000,
	"MTC/A,NHit100Med,dcOffset": 4900,
	"MTC/D,GtMask": 33580617,
	"MTC/D,CoarseDelay": 150,
	"MTC/A,NHit100Med,mV/Hit": -10,
	"MTC/D,PEDCrateMask": 33554431,
	"MTC/D,FineDelay": 0,
	"MTC/A,NHit100Lo,Threshold": 4020,
	"MTC/A,OWLELo,dcOffset": 4900,
	"MTC/A,ESumLow,mV/pC": -10,
	"MTC/A,OWLEHi,Threshold": 3000,
	"MTC/A,NHit100Hi,mV/Adc": 1,
	"MTC/A,ESumLow,Threshold": 1000,
	"MTC/A,OWLN,Threshold": 3000,
	"MTC/A,OWLN,mV/Hit": -35,
	"MTC/A,OWLEHi,dcOffset": 4900,
	"run_type_word": 1,
	"PED_PGT_Mode": false,
	"MTC/A,ESumLow,dcOffset": 4900,
	"MTC/A,NHit100Lo,mV/Adc": 1,
	"MTC/A,NHit20LB,Threshold": 3000,
	"MTC/A,NHit20,dcOffset": 4900,
	"MTC/A,OWLELo,mV/pC": -10,
	"Comments": "Nothing Noted",
	"MTC/A,ESumHi,mV/pC": -10,
	"MTC/A,NHit100Hi,Threshold": 0,
	"MTC/A,OWLELo,mV/Adc": 1,
	"MTC/D,Nhit100LoPrescale": 100,
	"XilinxFilePath": "/Users/snotdaq/Desktop/orca_files/MTC Xilinx Load",
	"MTC/D,ControlMask": 0,
	"MTC/D,PedestalWidth": 40,
	"MTC,tub": 40,
	"MTC/A,NHit20,mV/Adc": 1
}
```

The NaN value for the 10 MHz clock is not valid json.

tlatorre-uchicago, 2016-12-10:
The NaN was due to that field not being set in the MTC GUI. I've set it so now we can save standard runs. It might be good to just grab the thresholds and pulser rate.

fbdescamps, 2016-12-18:
Is this now resolved? @jcaravaca @tlatorre-uchicago

jcaravaca, 2016-12-20:
Not a problem anymore but it is not solved at code level, so it could strike again in the future. I'd leave it open as a reminder to look into this

tlatorre-uchicago, 2017-03-02:
Fixed in #274.

closed 2017-03-02 by tlatorre-uchicago