snoplus/orca #154: ORCA crash when quitting experiment - suspected ELLIE issue

fbdescamps, 2016-08-27:
To reproduce:
 - have ORCA sno+ running
 - save and close experiment
 - enjoy the beachball

Reported error:
```
2016-08-27 14:23:28.815 Orca[21101:548871] Failed to connect (vtMatrix) outlet from (ORFecDaughterCardController) to (NSMatrix): missing setter or instance variable
2016-08-27 14:29:31.231 Orca[21101:548871] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1343.16/AppKit.subproj/NSCell.m:1622
2016-08-27 14:29:31.241 Orca[21101:548871] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1343.16/AppKit.subproj/NSCell.m:1622
2016-08-27 14:29:31.258 Orca[21101:548871] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1343.16/AppKit.subproj/NSCell.m:1622
```

Screenshot

[image]

EdLeming, 2016-08-31:
@fbdescamps. Thanks for the spot. This is a easy fix - see PR #156

tlatorre-uchicago, 2016-09-22:
Fixed in #156.

closed 2016-09-22 by tlatorre-uchicago