snoplus/orca #155: Error contacting database when starting ORCA

fbdescamps, 2016-08-29:
These messages show up in the ORCA log when starting up ORCA:

'''
16-08-29 09:25:15 Opened [~/Desktop/snot_main.Orca]
16-08-29 09:25:20 Error contacting database: The request timed out.
16-08-29 09:25:20 Standard Run "" does not exist in DB. 
16-08-29 09:25:22 Error contacting database: The request timed out.
16-08-29 09:25:22 Standard Run Version "" does not exist in DB. 
16-08-29 09:25:24 Couldn't retrieve SR DEFAULT values. Error querying couchDB, please check the connection is correct. Error: 

--- error message ----

16-08-29 09:25:24 Standard Run "" does not exist in DB. 
16-08-29 09:25:25 Standard Run Version "" does not exist in DB. 
16-08-29 09:25:25 Cannot display DEFAULT values. There was some problem with the Standard Run DataBase. 
16-08-29 09:25:25 Cannot display TEST RUN values. There was some problem with the Standard Run DataBase. 
16-08-29 09:25:25 Standard Run Version "" does not exist in DB. 
16-08-29 09:25:25 Cannot display DEFAULT values. There was some problem with the Standard Run DataBase. 
16-08-29 09:25:25 Cannot display TEST RUN values. There was some problem with the Standard Run DataBase.

jcaravaca, 2016-10-25:
The SNO+ GUI was querying the DB automatically when starting ORCA and displaying these errors. This is not true anymore in the new PR - the SR will be updated upon user request by clicking a button, so this issue should be resolved once we merge it.

tlatorre-uchicago, 2016-12-17:
This issue has been resolved.

closed 2016-12-17 by tlatorre-uchicago