snoplus/orca #165: HV thread doesn't stop when experiment is closed

tlatorre-uchicago, 2016-12-01:
When closing the experiment in ORCA to switch to a new .Orca file, sometimes the ORXL3Model segfaults in readHVStatus. Here is the debugger output:

```
2016-12-01 16:54:08.301 Orca[49818:303] Invalid color System, labelColor (warning given only once)
2016-12-01 16:54:08.589 Orca[49818:303] Could not find image named 'CameraPic'.
2016-12-01 16:54:33.463 Orca[49818:303] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1265.21/AppKit.subproj/NSCell.m:1637
2016-12-01 16:54:33.466 Orca[49818:303] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1265.21/AppKit.subproj/NSCell.m:1637
2016-12-01 16:54:33.522 Orca[49818:303] *** Assertion failure in -[NSTextFieldCell _objectValue:forString:errorDescription:], /SourceCache/AppKit/AppKit-1265.21/AppKit.subproj/NSCell.m:1637
2016-12-01 16:54:34.293 Orca[49818:1c61f] -[__NSCFString crateNumber]: unrecognized selector sent to instance 0x2a00f0b0
2016-12-01 16:54:34.294 Orca[49818:1c61f] An uncaught exception was raised
2016-12-01 16:54:34.294 Orca[49818:1c61f] -[__NSCFString crateNumber]: unrecognized selector sent to instance 0x2a00f0b0

2016-12-01 16:54:34.304 Orca[49818:1c61f] (
	0   CoreFoundation                      0x907f0471 __raiseError + 193
	1   libobjc.A.dylib                     0x9904b091 objc_exception_throw + 162
	2   CoreFoundation                      0x907f4cb3 -[NSObject(NSObject) doesNotRecognizeSelector:] + 275
	3   CoreFoundation                      0x90740522 ___forwarding___ + 1010
	4   CoreFoundation                      0x9074010e _CF_forwarding_prep_0 + 14
	5   Orca                                0x00889440 -[ORXL3Model crateNumber] + 48
	6   Orca                                0x0089b2ef -[ORXL3Model readHVStatus] + 239
	7   Orca                                0x008a6f04 -[ORXL3Model(private) _hvXl3] + 4948
	8   Foundation                          0x92c3916e -[NSThread main] + 45
	9   Foundation                          0x92c390c6 __NSThread__main__ + 1426
	10  libsystem_pthread.dylib             0x933085fb _pthread_body + 144
	11  libsystem_pthread.dylib             0x93308485 _pthread_struct_init + 0
	12  libsystem_pthread.dylib             0x9330dcf2 thread_start + 34
)
2016-12-01 16:54:34.304 Orca[49818:1c61f] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString crateNumber]: unrecognized selector sent to instance 0x2a00f0b0'
*** Call stack at first throw:
(
	0   CoreFoundation                      0x907f0471 __raiseError + 193
	1   libobjc.A.dylib                     0x9904b091 objc_exception_throw + 162
	2   CoreFoundation                      0x907f4cb3 -[NSObject(NSObject) doesNotRecognizeSelector:] + 275
	3   CoreFoundation                      0x90740522 ___forwarding___ + 1010
	4   CoreFoundation                      0x9074010e _CF_forwarding_prep_0 + 14
	5   Orca                                0x00889440 -[ORXL3Model crateNumber] + 48
	6   Orca                                0x0089b2ef -[ORXL3Model readHVStatus] + 239
	7   Orca                                0x008a6f04 -[ORXL3Model(private) _hvXl3] + 4948
	8   Foundation                          0x92c3916e -[NSThread main] + 45
	9   Foundation                          0x92c390c6 __NSThread__main__ + 1426
	10  libsystem_pthread.dylib             0x933085fb _pthread_body + 144
	11  libsystem_pthread.dylib             0x93308485 _pthread_struct_init + 0
	12  libsystem_pthread.dylib             0x9330dcf2 thread_start + 34
```

fbdescamps, 2016-12-01:
@BenLand100 ?

BenLand100, 2016-12-01:
Does anyone know if there a notification I can listen for to stop the thread?

tlatorre-uchicago, 2016-12-01:
Whenever the XL3 connects/disconnects you can listen for a notification called XL3_LinkConnectionChanged (see https://github.com/snoplus/orca/blob/reset_crate/Source/Objects/Custom%20Hardware/SNO%2B/XL3/XL3_Link.m#L34). I'm not sure what the sequence of events is when the experiment is closed, but that might be enough.

BenLand100, 2016-12-02:
Unfortunately I already have a condition to exit the thread if the XL3 disconnects https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO%2B/XL3/ORXL3Model.m#L4791

It must be the case that you only see this when the objects get cleaned up before it calls readhvstatus but after it's checked that the XL3 is connected... I'm not super concerned about this being a dangerous bug but it's certainly not pretty. I'll see if I can find a more robust way to handle ORCA cleaning up the experiment - presumably it informs the objects before simply deleting them.

tlatorre-uchicago, 2017-01-22:
Fixed in #243.

tlatorre-uchicago, 2017-02-15:
Still seem to be having problems. I've attached a screenshot from the debugger when quitting Orca.
[image]