snoplus/orca #167: orca crash due to uncaught exception in _hvInit

tlatorre-uchicago, 2016-12-02:
After stopping and starting the XL3 server when ORCA reconnected to the XL3 server there was an uncaught exception in _hvInit. I've attached the debugger output:

```
2016-12-01 16:59:03.627 Orca[49900:303] Invalid color System, labelColor (warning given only once)
2016-12-01 16:59:03.811 Orca[49900:303] 10.4-style NSDateFormatter method called on a 10.0-style formatter, which doesn't work. Break on _NSDateFormatter_Log_New_Methods_On_Old_Formatters to debug. This message will only be logged once.
2016-12-01 16:59:04.020 Orca[49900:303] Could not find image named 'CameraPic'.
2016-12-01 17:01:55.451 Orca[49900:ef2b] ERROR: unable to get the receiver data from the DB!
2016-12-01 17:01:58.502 Orca[49900:1f453] ERROR: unable to get the receiver data from the DB!
2016-12-01 20:08:27.232 Orca[49900:1bf2b] An uncaught exception was raised
2016-12-01 20:08:27.232 Orca[49900:1bf2b] forwardInvocation:: NSUndoManager 0x2a244200 received forwarded invocation while invocation target is nil. Call prepareWithInvocationTarget: before invoking setHvNominalVoltageB:
2016-12-01 20:08:27.234 Orca[49900:1bf2b] (
	0   CoreFoundation                      0x907f0471 __raiseError + 193
	1   libobjc.A.dylib                     0x9904b091 objc_exception_throw + 162
	2   CoreFoundation                      0x907f038b +[NSException raise:format:] + 139
	3   Foundation                          0x92c7546f -[NSUndoManager _forwardTargetInvocation:] + 163
	4   Foundation                          0x92c753c0 -[NSUndoManagerProxy forwardInvocation:] + 43
	5   CoreFoundation                      0x907402f9 ___forwarding___ + 457
	6   CoreFoundation                      0x9074010e _CF_forwarding_prep_0 + 14
	7   Orca                                0x0088873f -[ORXL3Model setHvNominalVoltageB:] + 191
	8   Orca                                0x008a4bc1 -[ORXL3Model(private) _hvInit] + 801
	9   Foundation                          0x92c3916e -[NSThread main] + 45
	10  Foundation                          0x92c390c6 __NSThread__main__ + 1426
	11  libsystem_pthread.dylib             0x933085fb _pthread_body + 144
	12  libsystem_pthread.dylib             0x93308485 _pthread_struct_init + 0
	13  libsystem_pthread.dylib             0x9330dcf2 thread_start + 34
)
2016-12-01 20:08:27.234 Orca[49900:1bf2b] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'forwardInvocation:: NSUndoManager 0x2a244200 received forwarded invocation while invocation target is nil. Call prepareWithInvocationTarget: before invoking setHvNominalVoltageB:'
*** Call stack at first throw:
(
	0   CoreFoundation                      0x907f0471 __raiseError + 193
	1   libobjc.A.dylib                     0x9904b091 objc_exception_throw + 162
	2   CoreFoundation                      0x907f038b +[NSException raise:format:] + 139
	3   Foundation                          0x92c7546f -[NSUndoManager _forwardTargetInvocation:] + 163
	4   Foundation                          0x92c753c0 -[NSUndoManagerProxy forwardInvocation:] + 43
	5   CoreFoundation                      0x907402f9 ___forwarding___ + 457
	6   CoreFoundation                      0x9074010e _CF_forwarding_prep_0 + 14
	7   Orca                                0x0088873f -[ORXL3Model setHvNominalVoltageB:] + 191
	8   Orca                                0x008a4bc1 -[ORXL3Model(private) _hvInit] + 801
	9   Foundation                          0x92c3916e -[NSThread main] + 45
	10  Foundation                          0x92c390c6 __NSThread__main__ + 1426
	11  libsystem_pthread.dylib             0x933085fb _pthread_body + 144
	12  libsystem_pthread.dylib             0x93308485 _pthread_struct_init + 0
	13  libsystem_pthread.dylib             0x9330dcf2 thread_start + 34
)
(lldb) 
```

tlatorre-uchicago, 2016-12-06:
@BenLand100 It looks like this is an issue where the HV thread is calling a setter which calls the undo manager.

Are all of these setters only called by the HV thread? If so, since there is no user input I think the undo manager can just be deleted.

For any setters that are also called by the main thread and/or by GUI actions, then I think the only way to handle it is with a lock, and then just don't call the setter from the HV code.

Any chance you have enough time to look at this tonight? If not, I can try and go through. I think it's something we should try to fix before tomorrow otherwise there ORCA might crash while they are trying to ramp tomorrow.

BenLand100, 2016-12-06:
@tlatorre-uchicago The setHvNominalVoltage* were the only existing methods I used and for whatever reason they had undo manager calls in them. I can open a PR to fix in a moment but yes the fix is to remove the undo manager calls. Nothing else should call those methods besides the HV thread.

BenLand100, 2016-12-06:
Fixed in #174

closed 2016-12-06 by BenLand100