snoplus/orca #180: There should be  a standard run that changes no settings at all when started.

marzece, 2016-12-06:
I think the SNO term was "diagnostic" run.

jcaravaca, 2016-12-06:
@marzece what is the use case for this? In principle you won't be allowed to change settings in this run
since that'd be Maintenance.

marzece, 2016-12-06:
Could there be a special version of Maintenance that won't change any settings?

tlatorre-uchicago, 2016-12-06:
I think you should be allowed to change detector settings for this run. The idea is we need a way from the SNO+ run control to start new runs without touching the detector to debug things. I think we discussed this at the last DAQ workshop, and assigned bit 6 to this run.

jcaravaca, 2016-12-06:
#183 includes the diagnostic run. I'm not still convinced we need both Maintenance and Diagnostic, but for the time being I added that posibility and we will learn from experience whether both are actually useful.

tlatorre-uchicago, 2016-12-17:
Fixed in #183.

closed 2016-12-17 by tlatorre-uchicago