snoplus/orca #182: XL3 reporting nonsense values for HV after power on

BenLand100, 2016-12-06:
Seen on the teststand after powering up. I'm not 100% sure what's been changed in the XL3 initialization, but it seems like prior to doing anything in ORCA the XL3 reports that it is initialized, the supply is OFF, but that the readback is a very large value with a below-nominal current. Could we double check this on the XL3 / ml403 side? @tlatorre-uchicago @bonventre

[image]

tlatorre-uchicago, 2016-12-06:
There haven't been any changes to the ML403 initilization at boot up. I think this is an issue we've seen before. See #150.

bonventre, 2016-12-06:
Can you try two things: read TEST_REG and make sure it is set to 0xFFFFFFFF, and turn HV on to 0V and see if the readback is ok. Was the HV cable plugged in to a supply when this happened?

BenLand100, 2016-12-06:
Yep this does look like #150 and we probably would have gotten the same chain of events had we pressed HV ON. Trying to read the test register (or any other register) via orca gives a "command error" with a reason "xl3 bus error" but right now the test stand is in this state if anyone has any ideas?

tlatorre-uchicago, 2016-12-07:
Closing since same as #150.

closed 2016-12-07 by tlatorre-uchicago