snoplus/orca #187: HW Wizard should send PEDs to channels that have resistor pulled

fbdescamps, 2016-12-07:
We think we should allow to send peds to channels that have pulled resistor.

tlatorre-uchicago, 2016-12-17:
This is easy enough to change. We should probably talk to Phil and see if there is any good reason that we shouldn't allow pedestals on channels that have their resistor pulled.

tlatorre-uchicago, 2017-01-31:
I talked to Phil and he said that our current plan is that for PMTs with resistors pulled we are going to disable the sequencer and triggers and so it doesn't make a lot of sense to enable pedestals.

@fbdescamps Can we close this?

fbdescamps, 2017-01-31:
I disagree: we should have sequencers on for channel with pulled resistors if possible and therefore we should be able to send pedestals to them. Quote from Noel from an email discussion a while back:

```
For SNO, we never disabled the sequencer by default when pulling a
resistor, want to keep the digital end live which helps in understanding
crosstalk and to cut flashers. But sometimes you do want to disable the
sequencer when you pull a channel, ie. due to digital issues.
```

tlatorre-uchicago, 2017-01-31:
What does pull a channel mean? We should probably bring this back up on the next DAQ call.

tlatorre-uchicago, 2017-08-30:
Now that we have the channel database implemented, we need to update the hardware wizard to use the new channel database instead of the old pmt_info table. This should be fairly trivial to update at the same time.