snoplus/orca #206: Diagnostic-run resync or stop/start loads MTCA mask?

fbdescamps, 2016-12-16:
We noticed, when resync or stop/start diagnostic run, there is a burst of triggers. However, for the diagnostic run, the MTCA mask should not be touched? @jcaravaca Also, I do not fully understand why there is a burst of triggers when the MTCA mask is loaded?

jcaravaca, 2016-12-16:
The SR do not touch the MTCA masks. Also I thought the burst is happening always, not only with the diagnostic runs. If you want to narrow down the problem you can use the old control panel which do not load settings from DB. If you see the same problem, then the issue is at the start/stop run side.

tlatorre-uchicago, 2016-12-17:
The MTCA masks are reloaded on all hard run starts which includes resync runs, although for diagnostic runs the values shouldn't change. We could change this for diagnostic runs and really have no hardware being loaded at the start of the run. That's probably a good idea, but it'll take adding some logic to the SNO+ model functions which handle run starts.

fbdescamps, 2016-12-17:
OK, thanks! In this case, we can probably close this issue, it is nothing critical.

tlatorre-uchicago, 2016-12-17:
On run starts the global trigger mask is explicitly cleared before setting the MTCA+ thresholds to prevent this. I didn't think that setting the relays would also cause a burst but I can fix it so that it sets up the relays before loading the GT mask too.

tlatorre-uchicago, 2016-12-17:
Fixed in #215.

closed 2016-12-17 by tlatorre-uchicago