snoplus/orca #211: ping crates at run start

tlatorre-uchicago, 2016-12-17:
Should we ping all the crates on every run start?

@nbarros brought up the idea of doing this as part of a pre-run start procedure. It should be possible to do this during the beginning of a run when hardware is being loaded so that it occurs before the first valid GTID. This is only possible during hard run starts when the crates are initialized, so this means this won't be available for rollover runs or continuous run starts.

The only downside I can think of is that it is going to add some dead time to hard run starts (although not much) and that it might be annoying if we always do this.

fbdescamps, 2016-12-18:
Could you elaborate on what the reasons for adding this are? @tlatorre-uchicago @nbarros

tlatorre-uchicago, 2016-12-18:
I think it's to make sure that you are getting triggers from all the crates.

fbdescamps, 2016-12-18:
Thanks. To confirm: this would then be looked at by a DQ processor after the run. Then, the operator would be alerted of any issue when she/he looks at the DQ run-page to find out if a run passed preliminary DQ?

On a related note: it might be nice to have the 'ping crates' button do a much slower version, so that we can follow along and see any issues real-time.

Also: pinging does not change the channel trigger mask or crate trigger mask, right?

tlatorre-uchicago, 2017-01-10:
No, I don't know of any plan to have the DQ processor look at this. It would just be useful if you were going back to look at a ZDAB file you could easily see at the start of the run which crates were successfully sending triggers.

Hopefully with the update to the redisptacher it shouldn't be necessary to do this slower, you should be able to go back in the history and see all of the crates. This still needs to be tested.

No, pinging does not change any of the trigger settings, it just sets the pedestal mask.

tlatorre-uchicago, 2017-01-18:
Another benefit to this is that it would allow you to see quickly if a crate was out of sync.

fbdescamps, 2017-01-18:
Ok, after the discussions on this, I feel convinced that this is a good idea. Could you give the precise details of the implementation:
 * how many peds/crates
 * which ped mask
 * what rate

Is there any risk (high rate, all channel triggers on, full crate in mask), worst case scenario? Thanks!

tlatorre-uchicago, 2017-01-18:
The current ping crates just enables pedestals for the entire crate and cycles through all 19 crates once. There is no rate since it just fires one event per crate.

I don't think there is any risk to enabling this.

fbdescamps, 2017-01-18:
Ok, I understood that  it was a series of peds per crate. Once per crate sounds fine. Thanks!

tlatorre-uchicago, 2017-01-18:
The only outstanding question is exactly when in the run sequence we fire this. We could fire them in the "dead" time between the first and valid GTID or just after the valid GTID. Also, do we fire them on every run start or only on hard run starts?

marzece, 2017-01-18:
Perhaps we'd only want it at the start of physics runs. Or maybe just not at the start of maintenance/diagnositic runs. I can imagine after some break down or something  it'd be really annoying to switch to a new run and have my high nhit xsnoed flooded by 20 more unneeded events.

tlatorre-uchicago, 2017-01-19:
If this is going to be used to check for out of sync crates I think it should happen during any run start. The XSNOED history has a history of 100 so it wouldn't completely flood the high nhit XSNOED. We could also make sure to set up these XSNOEDs so they ignore any triggers with the pedestal bit set.

fbdescamps, 2017-01-21:
I am going to suggest to implement this at the start of all runs, if it turns out to be too annoying or problematic, we can dial it down to physics-runs only. We should do it after the validGTID: they will be tagged (PED) so there is no confusion and this way, we allow valid physics (we will be ignoring all GTID's before validGTID) to happen in the short amount of time that we need to ping the crates. ❔

tlatorre-uchicago, 2017-06-27:
This was implemented in #406.

closed 2017-06-27 by tlatorre-uchicago