snoplus/orca #216: Pedestals at high rate freeze up ORCA

fbdescamps, 2016-12-17:
When running pedestals at high rate, the latency increased dramatically 1/5-ms to about 20ms and ORCA becomes really sluggish. To reproduce: just set the pulser at 5kHz (no channel triggers) and try changing it back to 10Hz.

This is concerning since it tells us that ORCA will also become sluggish during a breakdown.

fbdescamps, 2016-12-18:
Is the solution here to remove the HV monitoring from ORCA and do it with an external piece of code? Maybe not have the heartbeats running for now?

tlatorre-uchicago, 2016-12-18:
I never actually confirmed that it was the database commands that were slowing down the main thread, but it seems like the only thing that is continuously sending requests underground.

I think it's probably worth first investigating exactly what the cause of the problem is, and if there is an easy fix in ORCA.

tlatorre-uchicago, 2017-01-18:
I'm unable to reproduce this issue on operag1 right now. I have tried:

- running the pulser at 10 kHz which should saturate the amount of data we can read out since it is dominated by the CAEN data

- running pedestals at 5 kHz and the pickup caused all the FEC FIFOs to overflow

In both cases ORCA was still very smooth. I tried running both normally and in the debugger and in both cases it was still very smooth.

fbdescamps, 2017-01-18:
Ah, well. That's great then :).  I wonder if this was because we were running xsnoed connected to Minard still at that time. Did you change the dispatcher settings at all?

tlatorre-uchicago, 2017-01-18:
I thought I remember seeing it after we had fixed both the frequency of the database requests and switched to using just the above ground dispatcher, but I'm not 100% sure.

The only other variable could be that we were noticing huge latency spikes every hour or so before, which we believe was when DEAP was copying their run data above ground.

fbdescamps, 2017-01-21:
I think the conclusion is here that we will try to implement a network-latency monitor to find correlations/problems. I'll be closing this for now since the high-rate pedestals were shown not to be the underlying cause of the freeze-up.

closed 2017-01-21 by fbdescamps