snoplus/orca #222: Nominal Voltage shown for all crates in HV Master in Control Panel

ecaden, 2016-12-20:
Instead of having the nominal voltage shown only when the rack is on, each crate's nominal voltage should always be shown on the HV master tab. Maybe some color change is appropriate if the rack is then on.

BenLand100, 2017-01-11:
@jcaravaca on the HV master tab would it be difficult to add logic to display a nominal voltage of 0 as 'OFF' or 'D/C' or 'N/A' and @ecaden would that satisfactorily solve this issue? The reason it's showing up as 0 now is the code to check the DB happens as the XL3's connect to avoid flooding queries right when orca starts - so crates that are off never query the DB.

ecaden, 2017-01-11:
The nominal voltage of each rack shouldn't change, I'm not sure why this is coming from the database? I would think it should be hard coded, also as the limiting voltage that each crate could ramp to.

BenLand100, 2017-01-12:
I have heard elsewhere that the nominal voltages are not set in stone - mostly because PMTs have been swapped out and there is a variation in base resistance. Since it's something that could potentially change having it compiled into ORCA is less than ideal because there's not really a substantial difference in having it hardcoded vs read from the database except that the former requires a recompile to change it. Maybe this is something to bring up on a DAQ call.

tlatorre-uchicago, 2017-01-22:
Fixed in @243.

closed 2017-01-22 by tlatorre-uchicago