snoplus/orca #228: Does not correctly set XL3 slot mask when setting XL3 mode.

bonventre, 2017-01-04:
It appears like the XL3 slot mask is set to slots present regardless of the slot mask set in the XL3 window. It is also not clear from the gui that this is only set when the mode is set to normal.

tlatorre-uchicago, 2017-01-08:
This was an intentional change, but maybe we need to revisit it. The problem was that the set XL3 mode button is very far away from the slot mask and most people did not really understand that when you set the XL3 to normal mode it only reads out those channels masked in above which were often different from the cards present because those same check marks are used to set the pedestal mask.

Not sure what the best solution is here.

tlatorre-uchicago, 2017-01-20:
Fixed in #254.

closed 2017-01-20 by tlatorre-uchicago