snoplus/orca #231: Handle base current/cmos rate polling outside of orca

marzece, 2017-01-06:
The amount of data coming from polling at 1hz for an entire crate puts a decent amount of stress on the computer. We should move the polling to somewhere else since no one wants to look at the output in the log anyways.

fbdescamps, 2017-01-06:
I think this could fit with the FEC voltages and temperature monitoring that Teal might work on. @pershint

tlatorre-uchicago, 2017-01-10:
I think the only thing that could reasonable cause the computer to run slower is the logging which IIRC takes a non trivial amount of time. It would be nice to confirm that with verbose output disabled you don't notice any difference.

tlatorre-uchicago, 2017-01-18:
I think Eric and I tested this and determined that it was indeed the logging that is slowing down ORCA here. Short term solution is to use the monitoring webpage which isn't ideal as it has a bit of latency or to just make sure not to poll too fast.

fbdescamps, 2017-01-18:
Do we still feel that we should handle the polling outside of ORCA and is @pershint working on this in addition to the FEC voltages and temperatures?

tlatorre-uchicago, 2017-01-18:
Not 100% sure but Billy expressed some interest in working on this.

marzece, 2017-01-18:
I think for sure we should remove a lot of the code that does polling and move that to the xl3 server. Doing that with the cmos rates might be tricky but I think totally doable.

mastbaum, 2017-08-30:
This is addressed by the check_rates script.

closed 2017-08-30 by mastbaum