snoplus/orca #232: Orca crash while enabling triggers in HW wizard

marzece, 2017-01-06:
This happened while we were enabling triggers on a crate after ramping it to HV. There was no good log messages of what exactly went wrong but the deubbger said a bad access happened at line [1717 of ORFec32Model.m](https://github.com/snoplus/orca/blob/master/Source/Objects/Custom%20Hardware/SNO+/Fec32/ORFec32Model.m#L1717)

We were turning triggers on for a crate that was missing a card so that may or may not have something to do with it.

fbdescamps, 2017-01-06:
Was the card also missing in ORCA?

marzece, 2017-01-06:
Yeah, we may have deleted just a 10 minutes earlier or something like that though.

tlatorre-uchicago, 2017-01-22:
@marzece If you see this again, could you post the steps to reproduce it? I've tried deleting cards on the teststand and running the HW wizard, but I can't seem to reproduce it.

tlatorre-uchicago, 2017-03-13:
@boardhead, you mentioned a while ago that you fixed a bug in the detector state loading code for missing boards, is that right?

closed 2017-04-14 by tlatorre-uchicago