snoplus/orca #233: ORCA does not save itself automatically

fbdescamps, 2017-01-06:
We need it to save:
 * at the start of each run
 * every 5min

fbdescamps, 2017-01-06:
@BenLand100 pointed out that this might be because of the fact that we use our own run control...

BenLand100, 2017-01-06:
That does look to be the case - see this line in the orca run control
https://github.com/snoplus/orca/blob/252a0bac2025b8d0bfd7cd3cc6fec6635a1181ae/Source/Objects/Control/RunControl/ORRunController.m#L666
which seems to be missing an analogous statement in our run control
https://github.com/snoplus/orca/blob/986b1630d03c4603be38c6714eb707903dab2210/Source/Experiments/SNOP/SNOPController.m#L411

fbdescamps, 2017-01-11:
@jcaravaca Any update on this?

fbdescamps, 2017-01-16:
@jcaravaca I think the latest here was that ORCA only saves to the current open file at run-start when the GUI actually changed. Has this been confirmed, and does ORCA indeed save at all types of run-starts when the GUI has changed? If so, please close this issue.

We will not be automatically saving every 5minutes or so due to the following arguments that were made against:
 * Experts sometimes use the unsaved state to be able to recover from changes that need to be reversed. This is a handy thing to keep for now.
 * If it done every run.. then you know the state. As in: ORCA crashes.. I wonder if that last change I  made was saved or not? That question is much easier if the save happens only at run boundaries.

tlatorre-uchicago, 2017-01-18:
Confirmed that ORCA does not save the .Orca file when you start a run from the SNO+ control panel.

fbdescamps, 2017-01-18:
Did you change something in the GUI before the run start?

tlatorre-uchicago, 2017-01-18:
Yes, we changed the PED -> PULGT button in the MTC GUI.

tlatorre-uchicago, 2017-01-20:
Fixed in #258.

closed 2017-01-20 by tlatorre-uchicago