snoplus/orca #245: Fetching an ECAL needs to zero manual corrections

tannerbk, 2017-01-17:
Or else you are in danger of putting thresholds into the noise. I don't think its safe to rely on someone remembering to go to the hardware wizard to zero them.

tlatorre-uchicago, 2017-01-17:
Still unclear why loading the ECALs caused so many channels to trigger. A quick check loading ECALs on crate 0 did not show any funny behavior. Someone probably needs to:

- check the detector state before and after this was done to verify that the thresholds actually changed
- look at the ECAL loading code and document exactly what the behavior is

fbdescamps, 2017-01-17:
This seems to be related to issue #153, where we discussed the possibility of allowing both: 
 * fetch ECAL and zero all manual corrections
 * fetch ECAL without zeroing manual corrections.

tlatorre-uchicago, 2017-01-17:
We just checked and there was not a single channel whose threshold was increased after zeroing out the manual corrections, so it doesn't seem like this issue was caused by low thresholds.

tlatorre-uchicago, 2017-01-17:
For future reference here are the interesting run numbers:

12577 - before loading ECALs
12578 - after loading ECALs when N20 started triggering at 20 kHz
12587 - after zeroing manual corrections when everything was back to normal

tlatorre-uchicago, 2017-01-17:
We haven't been able to come up with a theory for how this happened. Looking at the stream page on minard however there is a ~30 minutes of no runs before and ~45 minutes of no run afterwards until it was fixed, so it's possible that these high noise events were uncorrelated with loading the ECALs.

tlatorre-uchicago, 2017-01-17:
And in addition the first time a smooth run came back up the MTCA relays had changed, and so there was an even bigger gap until things were smooth with all the crates back in the MTCA+ mask.

tlatorre-uchicago, 2017-01-17:
We loaded the entire detector state from the run when the detector was triggering at 2 MHz and didn't see anything, so this is not reproducible.

closed 2017-01-18 by tlatorre-uchicago