snoplus/orca #249: ORCA wouldn't save or load new test run after switching into expert mode

pershint, 2017-01-18:
On the graveyard shift on Jan. 18, 2016, we noticed that PED was masked into the MAINTENANCE, DEFAULT run.  After consulting with Freija, we switched from operator mode to expert mode to change PED to PGT.

We decided to try changing PED to PGT and saving as a new test run type (test run DEFAULTNOPED under run name MAINTENANCE).  ORCA tried to save the run as DIAGNOSTIC - DEFAULTNOPED unsuccessfully, and we could not load DEFAULTNOPED (see below).

[image]

We then tried to overwrite MAINTENANCE - DEFAULT with all the default settings, except PED -> PGT.  Similar errors printed when trying to save MAINTENANCE DEFAULT and re-load it (see below).

[image]

pershint, 2017-01-18:
Eric told us that this issue is due to having empty fields (specifically, the 10MHz clk (31..0) and bits (56..32) fields in the MTC/D panel).  After putting in zeroes, the save feature works.

marzece, 2017-01-18:
For the record this will be solved in a some what more permanent manner in my forthcoming MTC changes. After that we shouldn't have to worry about those fields being blank and screwing things up. We can leave this issue open till then though so I can get more issue closing stats when I submit the PR.

tlatorre-uchicago, 2017-01-23:
Dupe of #152.

closed 2017-01-23 by tlatorre-uchicago