snoplus/orca #250: unable to unlock the SNO+ control panel

tlatorre-uchicago, 2017-01-18:
Currently on operug1, it is impossible to unlock the SNO+ control panel. When you click on the lock, the computer beeps but no dialog appears.

I verified that the lock works in the FEC GUI, so it appears to be just that window.

tlatorre-uchicago, 2017-01-18:
I can't seem to reproduce this after closing and opening the SNO+ control window, so I'm not sure what caused the problem.

jcaravaca, 2017-01-19:
Could that be related with the fact that the panel was frozen?

tlatorre-uchicago, 2017-01-19:
No, the rest of the GUI was responsive.

jcaravaca, 2017-03-14:
@tlatorre-uchicago have you seen this again recently? I didn't hear about anyone complaining about this nor see this behavior myself, so I'll close the issue for now.

closed 2017-03-14 by jcaravaca