snoplus/orca #251: DB values not showing up in the SNO+ control panel

tlatorre-uchicago, 2017-01-18:
I was in a MAINTENANCE run, then selected PHYSICS run from the drop down list and hit overwrite standard run.

After this, the DB values column shows nothing for PHYSICS and MAINTENANCE runs, but still shows OK for all the other runs.

Here is the relevant messages from the Orca log:

```
17-01-18 15:22:29 Unversioned directory17-01-18 15:22:29 /Users/snoperator/Desktop/Orca.app/Contents/Resources/svnversion17-01-18 15:22:29 ---------------------------------------
17-01-18 15:22:29 Run 12637 started(cold start).
17-01-18 15:22:29 DataTaking Thread Started
17-01-18 15:33:47 Saving settings for Standard Run: PHYSICS - Version: DEFAULT ........ 
17-01-18 15:33:47 PHYSICS run saved as standard run. 
17-01-18 15:33:47 CouchDB
17-01-18 15:33:47 rev : 1-dbb1cd2d102fdc3acedb0e03ba61209e
17-01-18 15:33:47 ok : 1
17-01-18 15:33:47 id : 2c87d4a634c0685f1a16f1682cf8baa0
```

This issue persists after refreshing the standard runs.

tlatorre-uchicago, 2017-01-18:
I think this was caused by these standard runs having the diagnostic bit set in the run type word. Not sure how that happened, as I was definitely in a MAINTENANCE run before selecting PHYSICS run from the drop down menu. Perhaps last night shifters accidentally saved a maintenance run with the diagnostic bit set?

I think in the future we should not allow the operator to change the run type word for standard runs.

tlatorre-uchicago, 2017-01-18:
Maybe related to #253.

jcaravaca, 2017-01-19:
That'd be my guess: if the diagnostic bit is enabled the DB values are not shown. I see no errors in the log so that's the only explanation. Currently only experts can change the Run Type Word. This is because otherwise there is no way to properly set it, unless we change it directly in the DB. When creating or overwriting SR, the expert should follow a strict check list so that this does not happen. That'll be included in the detector manual.

closed 2017-01-23 by tlatorre-uchicago