snoplus/orca #252: DB values do not refresh after saving a standard run

tlatorre-uchicago, 2017-01-18:
After overwriting a standard run the DB values do not refresh until hitting refresh standard runs.

jcaravaca, 2017-01-19:
There is definitely a statement to do that after creating a SR and I'm not able to reproduce this locally. There might be a race condition since the DB query is async. Do you see this always on site?

tlatorre-uchicago, 2017-01-22:
I'm not able to reproduce it now. However, if I go to diagnostic run and then create a new standard run type neither the DB values or the test run show up until hitting refresh standard runs.

What is the race condition?

jcaravaca, 2017-01-23:
I can't reproduce it either but I suspect the problem might be here:
https://github.com/snoplus/orca/blob/master/Source/Experiments/SNOP/SNOPController.m#L1607
The query happens in parallel and it might take some time. So when you refresh the displayed values the new table might still not be in place.
I need to include a condition there to wait until the table is in the DB, which is not very easy at the moment.

tlatorre-uchicago, 2017-03-13:
Fixed in #331.

closed 2017-03-13 by tlatorre-uchicago