snoplus/orca #253: Multiple SNO+ control panels

tlatorre-uchicago, 2017-01-18:
Right now on operug1 there appear to be multiple SNO+ control panel windows.

[image]

This is a problem because they also seem to be inconsistent so it's not obvious which one actually corresponds to the SNO+ model.

jcaravaca, 2017-01-19:
Any recipe to reproduce this? I have never seen this behavior before

tlatorre-uchicago, 2017-01-19:
No, I just saw it when I took over during the crossover shift.

tlatorre-uchicago, 2017-02-08:
As @jcaravaca pointed out, someone probably pressed Command-C to make two windows. The fact that one was inconsistent was probably just #73.

closed 2017-02-08 by tlatorre-uchicago