snoplus/orca #260: ramping up crate 16 b produced a current near zero alarm

tlatorre-uchicago, 2017-01-19:
Not sure if this is just a value that needs to be adjusted. We are at 200 V and the current is 0.9 mA.

BenLand100, 2017-01-19:
@tlatorre-uchicago pretty sure this is just a value that needs adjusting. I believe it is set to 1mA for >= 100V

tlatorre-uchicago, 2017-01-19:
Yeah, this is probably because this supply only supplies HV to three cards instead of 16. We are keeping track of the HV settings and current values in this shift report: http://snopl.us/shift/view/06d9c7e5f1d640fd9ff7130994d7977d.

BenLand100, 2017-01-19:
Feel free to adjust the db as needed or I can do it later - just note that an Orca restart is necessary for db modifications to take effect.

BenLand100, 2017-01-21:
Lowered the threshold for 16B to 0.2 mA for >= 100V
Current readback is zero to one decimal place for supplies that are off so this should cause no issue.

closed 2017-01-21 by BenLand100